open-dynamic-robot-initiative/open_robot_actuator_hardware #2: motorware repo not public yet

jviereck, 2019-12-05:
For flashing the firmware on the TI boards, we rely on the `motorware` repo from git-amd right now. This means anyone outside of amd-clmc won't be able to flash this board.

Is this a problem?

\cc @luator @vincentberenz

luator, 2019-12-06:
We don't have a public fork of motorware due to an unclear legal situation (it is affected by some US export regulation...). However, we provide a patch with our modifications here: https://github.com/open-dynamic-robot-initiative/amd_motorware_ext. So people from outside have to download it from TI first and then can apply this patch.

luator, 2020-02-13:
Closing this as WONTFIX. Feel free to re-open if there is still something to discuss.

closed 2020-02-13 by luator