open-dynamic-robot-initiative/open_robot_actuator_hardware #4: motor shaft

f-e-l-i-x, 2019-12-14:
I have a few questions regarding the motor shaft: 
- why does the original have to be replaced (is it too short?)
- why does the replacement shaft have a different thickness?
Thanks

thomasfla, 2019-12-14:
Maybe @fgrimminger is the best person to answer, but here is some elements:

Yes, the motor shaft have to be replaced because it is too short. The longer one allows to insert a pulley, an encoder codewheel, and a support bearing. The thickness are the same, it's just that the original tolerances make the fit to tight to be reassembled without special tools, so we adjust the bore diameter with a reamer.

see https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/actuator_module_v1/details/details_motor_shaft_preparation.md
and 
https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/actuator_module_v1/details/details_motor_preparation.md
for more details;

fgrimminger, 2019-12-16:
The motor is designed for quadcopters - so the propeller is attached to the rotor from the top.
For our design it`s more convenient to extend the output shaft and attach the motor pulley from the opposite side. (see screenshot below) The motor shaft diameter is pretty much the same. For us it's easier to source stainless steel shafts with 3.98mm. That also helps with assembly because the pressfit on the rotor is less tight.
[image]

closed 2019-12-16 by fgrimminger