open-dynamic-robot-initiative/open_robot_actuator_hardware #10: quadruped_upper_leg

STGRobotics, 2020-05-11:
Thank you for the great work. Where can I find the quadruped_upper_leg base and cover.

Thank you

thomasfla, 2020-05-11:
Hi, Are you talking about the 12dof version? 
If yes, the upper leg actuator is the same as the one used for the 8dof. 
You can find the files here:

https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/actuator_module_v1/stl_files/actuator_module_shell_base.STL

and 

https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/actuator_module_v1/stl_files/actuator_module_shell_cover.STL

closed 2020-05-11 by STGRobotics

thomasfla, 2020-05-11:
@fgrimminger Maybe we want to add a link to the files in the .md?

STGRobotics, 2020-05-11:
It would be helpful. The name of the file through me for a loop.

fgrimminger, 2020-05-12:
I agree that it's a bit confusing.
I'll try to restructure the documentation to make it easier to use.
For now I have added the STL files to this folder.
https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/tree/master/mechanics/leg_3dof_v1/stl_files