open-dynamic-robot-initiative/open_robot_actuator_hardware #11: hip_aa_support/Bearing

STGRobotics, 2020-05-11:
Having problem finding the 20x25x4mm bearing locally. Is it posible to get the step file to modify for a different bearing size.

fgrimminger, 2020-05-12:
Yes - I uploaded the STEP file here:
https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/tree/master/mechanics/quadruped_robot_12dof_v1/step_files
Please note that the wires for the upper leg motor and encoder are routed through this part.
You'll want to make sure that the connectors still fit through the bearing and the support.
Also if you want to maintain the large range of motion the hip aa support  has to clear the upper leg rotor. 
[image]
[image]

STGRobotics, 2020-05-12:
Thank you for the file and post. I'm able to find 20x27x4mm bearing easily.

STGRobotics, 2020-05-12:
Found plastic bearing very affordable, to reduce the weight of the robot.

https://www.igus.com/product/460
B6704B3G 20x27x4mm  2.62 USD
B6705B3G 25x32x4mm  2.62 USD

fgrimminger, 2020-05-12:
Thanks - this is interesting....

closed 2020-05-14 by fgrimminger