open-dynamic-robot-initiative/open_robot_actuator_hardware #13: Hex Drive Flat Head Screws

STGRobotics, 2020-05-13:
I counter sink the body_structure_center, body_structure_side. Provides a larger grip surface area. Below are the flat heads I'm using. Would like the step file.

M2.5
https://www.mcmaster.com/91294a022-91294A022

M3
https://www.mcmaster.com/91294a134-91294A134

[image]

fgrimminger, 2020-05-14:
You can find the step files here now...
https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/tree/master/mechanics/quadruped_robot_12dof_v1/step_files

STGRobotics, 2020-05-14:
I meet to say, I modified the files and would you like to have them. Regardless thank you for uploading the step files.

STGRobotics, 2020-05-17:
Replacing metal screw with nylon slotted flat head screws to make bot lighter. Below link is where I purchased from.

https://www.mcmaster.com/plastic-screws/material~plastic/material~nylon-plastic/head-type~flat/thread-type~metric/length~16millimeters/

fgrimminger, 2020-05-18:
o.k. - thanks - please let us know if this modification works well for you.

closed 2020-09-02 by fgrimminger