open-dynamic-robot-initiative/open_robot_actuator_hardware #14: Custom Electronic Boards

STGRobotics, 2020-05-14:


thomasfla, 2020-05-14:
Hi, 
It could be a good idea! Unfortunately you will make a 1d sensor instead of 2d (lateral forces will not be perceived).
Also you will need some resistors to polarize the transistor and current limit the LED, so in the end it could be also quite complicate. I'm not sure about the weight of the OPB616 sensor, at the location, it is critical!
This foot sensor was design for solo8, but we are not sure about solo12 since the force really is 3d now.. We are trying to develop controller that do not need to contact sensor and estimator that can tell us if the foot is in contact.

STGRobotics, 2020-05-14:
Your correct did not take into consideration 2/3d, thank you.

STGRobotics, 2020-05-14:
How about using force sensing resistors (FSR) and IMU.

fgrimminger, 2020-05-18:
That should also be possible.
At the moment on each motor driver board there are two analog input channels available that are transmitted at 1kHz. One of them is used to measure the foot contact sensor signal. You can connect any other sensor with 0V-3.3V range.
Integrating the imu communication into the micro-controller would probably take quite a bit of effort.

closed 2020-09-02 by fgrimminger