open-dynamic-robot-initiative/open_robot_actuator_hardware #15: Dual Motor Testbed STL

fcaponetto, 2020-06-06:
Hi,

First, I would say thanks for all the effort that you put on this project. Especially, the well documented steps for the open source community!

I noticed that the "Dual Motor TestBed" project STL file is missing. Therefore, I would like to kindly ask you whether you can also share that project:

https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/raw/master/mechanics/dual_motor_testbed_v1/images/dual_motor_testbed_2.jpg

Thank you in advance.

fgrimminger, 2020-06-08:
Hi,
thanks for your interest.
You can find all the required STL files here now: https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/dual_motor_testbed_v1/README.md#3d-printed-parts

I also added an eDrawing assembly file that might be useful:
https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/dual_motor_testbed_v1/README.md#how-to-work-with-this

Best wishes.

fcaponetto, 2020-06-10:
Thanks a lot!

closed 2020-09-02 by fgrimminger