open-dynamic-robot-initiative/open_robot_actuator_hardware #16: Machined parts

njdjacobs, 2020-06-21:
The show-stopper for me is the requirement for machined parts. Not everyone has access to the necessary equipment to make these (or the skills to use it).

Can anyone suggest how to source these parts (preferably in Europe)?

fgrimminger, 2020-06-23:
You'll want to search for machine shops in your region and ask for quotes.
The machining of the pulleys becomes easier if you provide a custom form cutter for the tooth geometry. For that you'll first have to find a tool manufacturer that can produce the custom form cutter: https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/actuator_module_v1/details/details_machined_parts.md#option-a---form-cutter

Alternatively you could request a quote from an online manufacturing service.
We haven't tried this yet so we have no experience regarding the quality and cost - but we'll be looking into this topic soon. 
Here are some companies that we found:
https://www.rapiddirect.com/
https://www.facturee.de/en/about-us/

njdjacobs, 2020-06-23:
Thanks for the advice and links.

closed 2020-09-02 by fgrimminger