open-dynamic-robot-initiative/open_robot_actuator_hardware #17: Costs, sources

i-make-robots, 2020-06-30:
Hi!  Love the project.  Very interested in the encoders.

How much did they cost you?  An approximate idea of the build cost for the entire bot would be nice to have, and these encoders are of particular interest to me.

How did you find them in the first place?  I'm always looking for better sources.  Since your search-fu outdoes mine, I wish to learn your technique.

Thank you!

fgrimminger, 2020-07-06:
Hi! Thank you.
Yes - the encoders are nice - high resolution and very reliable.
I first saw the encoder being sold with Maxon and Faulhaber motors many years ago.
You can find the encoder kit using this product number: AEDM-5810-Z12
DigiKey and Mouser sell them for around 45€/piece.
Here is a short description on how to disassemble the kit:
https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/actuator_module_v1/details/details_encoder_kit_disassembly.md#details-encoder-kit-disassembly

The material cost for the Solo 8 robot is around 4000€  - for the Solo 12 robot around 6000€.

Here is another interesting encoder - it has the highest resolution that I could find for this size (80.000cpr!) but more complex electronics.  http://www.downloads-numerikjena.de/sheets/UM_ULTHEMIUS_eng.pdf

i-make-robots, 2020-07-06:
Nice!  And just to confirm... these are not absolute encoders, right?

fgrimminger, 2020-07-06:
Right - the Broadcom and the Ulthemius encoders are incremental encoders with an index pulse (ABZ).

closed 2020-09-02 by fgrimminger