open-dynamic-robot-initiative/open_robot_actuator_hardware #20: firmware

1027743011, 2020-07-14:
Hi! 
    Thank you for share this magic project.I love this project and i want  to make  a  quadruped robot 8dof v1  by myself.
    I  noticed the firmware  in the "open-motor-driver-initiative"   was temporarily hosted in this private repository: https://gitlab.laas.fr/open-dynamic-robot-initiative/omodri.  And  I can't connect to this website.This question has puzzled me for a long time.Therefore, I would like to kindly ask you whether you can also share that project.Thank you very much!

fgrimminger, 2020-07-15:
Hi,
thanks for your interest.

You can find instructions on how to download and patch the firmware here:
https://github.com/open-dynamic-robot-initiative/amd_motorware_ext#amd-motorware-extensions

The firmware is based on the Texas Instruments Motorware library.
We cannot make the modified firmware available, because TI requires you to confirm that you won't violate their export regulations before downloading Motorware.

1027743011, 2020-07-23:
HI,
thanks for your reply.I have successfully downloaded  MotorWare v1.01.00.18 from the TI website.But I can’t log in to the website where the source code is saved.Can this source code be published？I am a graduate student and I find it difficult to understand all the codes.So I want to ask if the source code will be published in other ways.
Thank you very much anyway!

jviereck, 2020-07-24:
Hi @1027743011 ,

I assume you are using the TI launchpads with CAN communication for your project? If so, the source code is open source. 

What you need to do is clone the following repos in the same folder:

- https://github.com/open-dynamic-robot-initiative/mw_dual_motor_torque_ctrl
- https://github.com/open-dynamic-robot-initiative/user_config_f28069m_drv8305
- https://github.com/open-dynamic-robot-initiative/amd_motorware_ext

This looks then like this:

```
workspace
├── amd_motorware_ext
├── mw_dual_motor_torque_ctrl
└── user_config_f28069m_drv8305
```

Then follow the [instructions](https://github.com/open-dynamic-robot-initiative/amd_motorware_ext) in the `amd_motorware_ext` to prepare the `motorware` folder.

Finally, you need to flash the TI board. For this, you can follow the documentation from the udriver boards. See the instructions [here](https://github.com/open-dynamic-robot-initiative/udriver_firmware/tree/master/firmware#installation--board-flashing).

Let me know if you have any questions left.

Best,
Julian

closed 2020-09-02 by fgrimminger