open-dynamic-robot-initiative/open_robot_actuator_hardware #21: Will the project be providing more detailed BOMs for some of the various mechanical configurations?

androiddrew, 2020-07-14:
Will the project be providing more detailed bill of materials for some of the various mechanical configurations? A huge challenge for me getting started with experimenting with a new design is sourcing all the materials. Some projects I have followed have provided a BOM for a standard configuration(with suppliers) which had me up and running in under a month. 

Even if this isn't added to the project page, could one of the maintainers be encouraged to provide maybe a blog post(or several) on a build?

easybuildtech, 2020-07-14:
I have been trying to source the machined parts.  Perhaps we can get a
handful of folks interested to commit in order to spread the setup costs
out?  The rest of the components I have been able to source relatively
effectively.

---
Benjamin R Smith
0bensmith0@gmail.com
c: 248 941 4022

fgrimminger, 2020-07-16:
Yes - we'll work on adding detailed bills of materials next.

Teaming up for ordering machined parts is a good idea.

We have setup a Discourse forum for questions and collaboration.
[https://odri.discourse.group/categories](url)

fgrimminger, 2020-07-20:
The bill of materials for the core components of the actuator module  can be found here now:
https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/actuator_module_v1/README.md#bill-of-materials---brushless-actuator-module-core

wujiang, 2020-11-20:
@easybuildtech did you manage to get those mechanical parts?

MusubaPy, 2021-02-09:
Your project is _amazing_! 

Can I use TMS320-XDS100-V3, JTAG instead of Spectrum Digital XDS100V2 JTAG?

Please advise what can be used instead of Lord Microstrain 3DM-CX5-25?

fgrimminger, 2021-02-10:
Thank you!
The TMS320-XDS100-V3 programmer should work but we haven't tested it.
I don't know about IMU alternatives - the master board electrical interface and firmware was designed for the Microstrain IMU.
If you want to use a different IMU you'll have to adapt the firmware and possibly the the electrical interface.
Alternatively you could use an external motion capture system for base estimation - that's what we've been using for a lot of the experiments in the past.
We also have a [Forum](https://odri.discourse.group/latest) now where you can ask questions and discuss alternative components.

wangyear, 2023-07-16:
Hi, my friends, I'm sorry to bother you to tell me that what the sensors have you used in your projects, and its concrete information.
Thank you very much, I'm looking forward to your reply.