open-dynamic-robot-initiative/open_robot_actuator_hardware #23: Identification of Torque-Current constant k_i

juanmed, 2020-08-13:
Hi

Thank you for sharing such an awesome robotics initiative.

According to your [paper in arxiv](https://arxiv.org/abs/1910.00093), the constant k_i relates the joint torque and the current of the motor. Since this parameter is a property of the motor, could you please comment on the experimental procedure to measure it? 

Thanks!

fgrimminger, 2020-08-17:
Hi,
we applied different currents to the motor and measured the output torque using a pulley, a string and a digital force meter.

Another option is to measure the velocity constant of the motor and calculate the torque constant from that.
Here is a good explanation: https://things-in-motion.blogspot.com/2018/12/how-to-estimate-torque-of-bldc-pmsm.html

juanmed, 2020-08-17:
Thanks for your reply.

closed 2020-08-17 by juanmed