open-dynamic-robot-initiative/open_robot_actuator_hardware #24: Gerber files for contact switch

Sreevis, 2020-09-08:
Hi, Thanks for this wonderful contribution.

I wish to make the 12 DOF robot configuration. 
I noticed a few things while checking the [foot contact switch PCB](https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/foot_contact_switch_v1/documentation/foot_contact_switch_v1.zip))  

[image]

The .brd file doesn't have copper pouring as shown in the Picture, sensor out track is broken, r6 is shorted .   and Wire strap pins which I supposed to be external to the PCB is in the .Brd file. 

Do you have the Gerber files for the foot contact switch?

fgrimminger, 2020-09-15:
Thanks for your feedback.
You are right - we must have uploaded the wrong version.
We are looking into it.

fgrimminger, 2020-09-30:
Hi,

Joshi Walzog has corrected the Eagle files and has provided the Gerber files for the foot contact switch.

You can find the files here now:
(https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/foot_contact_switch_v1/README.md#details-foot-contact-switch)

Sreevis, 2020-10-01:
Thank you Felix and Joshi. :+1:

closed 2020-10-01 by Sreevis