open-dynamic-robot-initiative/open_robot_actuator_hardware #25: Motor selection

juanmed, 2020-10-21:
Hi,

Thanks again for sharing this awesome project and design files. If possible, I would like to ask a rather general question: what was the design criteria to select the T- Motor Antigravity 4004, 300KV motor for the robot? The paper talks briefly on its torque characteristics only. Narrowing down, I would like to know more on your selection process, if there were any soft/hard requirements you wanted to satisfy, if you tested other motors, etc. 

Thanks for your time!

fgrimminger, 2020-10-22:
Hi,
thanks for your interest.

For this project we were looking for a high torque and low rotor inertia motor with a weight of around 50g.
Brushless outrunner motors with a high number of pole pairs and a low KV value are best suited.
We estimated that a torque constant of 0,02Nm/A would be required for high performance behaviors with a low gear ratio transmission of around 10:1 and motor drivers that can deliver up to 15A.
We looked at some hobby rc and quadcopter motors but didn't do much performance  testing.

Most motors we looked at were too heavy, too tall for our design or the KV value was too high.
We were interested in a motor with minimal height because that influences how much the leg segments need to be offset in this arrangement. (screenshot below)
The Antigravity motors have the perfect form factor, a torque constant of around 0,02Nm/A, are very high quality and can be modified easily to fit our design.
As a research lab that can only order components on invoice it was also important that the motor would be available for many years from resellers in Germany that would deliver on invoice.

At the time (2016) there were not so many choices available - today there might be better options.
Ben Katz has posted a comparison of different motors that he has tested:
[Motor Characterization](http://build-its-inprogress.blogspot.com/2016/01/motor-characterization-for-small.html)

Please let us know if you find a motor that you consider superior.
You can also use the[ forum](https://odri.discourse.group/categories) to discuss questions like this with us and others.

Best. Felix
[image]

juanmed, 2020-10-22:
Hi, thank you for the detailed answer and taking the time for the screenshot, this is very informative. Indeed, I am looking for other motors. Thanks for pointing out the discussion forum, I will for sure use it and let you know if I find anything worth considering.

closed 2020-10-22 by juanmed

fgrimminger, 2020-10-22:
Hi, 
you are welcome.
Good luck and best wishes,
Felix