open-dynamic-robot-initiative/open_robot_actuator_hardware #29: Connecting The Finger Optoswitch Connector

HanTop310, 2020-12-07:
Hi there, I need your help in proving me an idiot.
Can you please show me on the MASTER-BOARD where is the Finger Connector is connected?
Is it connected on the MICRO-DRIVER? Where?

fgrimminger, 2020-12-07:
Hi,
we connect the optical sensor to one of the analog-in ports of the micro driver board.
There are two analog-in ports available per board (ADC-A6 and ADC-B6).
Both analog measurements are transmitted through the masterboard to the control pc at 1kHz.
The pins are marked in the screenshot below - you could use a 3 pin connector for connecting one optical sensor or a 4 pin connector for connecting two optical sensors.
Make sure to use the 3.3V supply voltage - the analog inputs don't tolerate higher voltages.
Best. Felix
[image]

HanTop310, 2020-12-07:
Does it matter to which micro driver i connect? cause there are 6 micro drivers (in the 12dof) and 4 sensors

fgrimminger, 2020-12-08:
No -  it doesn't matter - you will just need to process the right sensor values from the data stream.
You can connect the optical sensors to any micro diver board.
All 12 analog-in channels are transmitted to the pc.
I would connect one sensor to each of the top two cards in the front and in the back.
That way you have separate 3pin connectors for each sensor - and the upper cards are easier to access.
We use the [90 degree Hirose connectors ](https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/electronics/details/details_components.md#hirose-df13-right-angle-pin-headers) on the micro driver board - that makes it easier to plug and unplug the sensors on the robot.

closed 2021-01-11 by fgrimminger