open-dynamic-robot-initiative/open_robot_actuator_hardware #36: Cable routing improvement proposal solo12

pamartinez-pal, 2021-05-19:
We are building a solo 12 in pal-robotics and we have found the cable routing around the shoulder joint very complicated. We had some issue with the cables being torn by the consecutive 90º forlds.

The assembly would be much simpler and durable if the cable channel was 3 to 5 mm wider. Would it be possible to expand the "Body Structure Hip AA Support" such that it has more room for the cabling. 

[image]
[image]

Thank you.

fgrimminger, 2021-05-20:
True - the internal wire passage is pretty tight.
I increased it by 3mm - from 4mm to 7mm.

You can find the updated STL file [here](https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/quadruped_robot_12dof_v1/stl_files/body_structure_hip_aa_support_rev_a.STL).
Does this solve your issue?

[image]
[image]

Best. Felix

pamartinez-pal, 2021-05-20:
Great!

Thank you.

closed 2021-05-20 by fgrimminger