open-dynamic-robot-initiative/open_robot_actuator_hardware #38: Cable Routing improving proposal

nitrostormbcn, 2021-05-28:
Hi, I'm Pau from PAL Robotics, we have found some improvement in the part "hip_fe_module_shell" that would greatly improve cable routing. We've had problems assembling the shoulder and this would ease manufacturing without changing the robot's geometry.

Before:
[image]

After: 
[image]

It only modifies the external shell and help routing. I can send you the modified part if you wish.
[image]
[image]
[image]
[image]

fgrimminger, 2021-05-28:
Hello Pau,

thanks for your suggestions.

I am surprised to hear that you have so much trouble with the wire routing through the hip.
Are you using the same encoder and motor phase connectors that we are using? ([Details Wiring](https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/electronics/details/details_wiring.md#details-wiring))

What kind of wires are you using?
We had bad experiences with ribbon wires.

I first pull the encoder connector and wires through - then the three motor phase wires - one after the other.
Using pliers helps sometimes.

Best. Felix

[image]

nitrostormbcn, 2021-05-31:
We've had problems with the cables getting twisted over each other and stripping the insulation, this should be alleviated with looser cable routing. 

Also when replacing the broken wires, disassembly and reassembly was very tedious.

Imho, I think this problems would be solved by allowing more space for the cables to pass.

fgrimminger, 2021-08-23:
Hi Pau,

o.k. - thanks for your suggestion and the stl file.
I added it to the [3dof leg documentation](https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/tree/master/mechanics/leg_3dof_v1#3d-printed-parts). 

Best,
Felix

closed 2021-08-23 by fgrimminger

dvogureckiy99, 2022-10-26:
good afternoon, thanks for the improvement, could you please provide a model in the extension .step or .SLDPRT, not .STL because all sizes are losted in it.