open-dynamic-robot-initiative/open_robot_actuator_hardware #42: what is the recommended power supply？

FranCOOlin, 2022-03-14:
I decide to use a switching power supply, but how much current it may consume during normal operation at the voltage of 24V ?

fgrimminger, 2022-03-21:
We use a 30A power supply - you can find more information here:
https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/electronics/details/details_components.md#power-supply

The current draw will depend on the gait of the robot - on average it's typically around 3-5A.
The problem with the power supplies isn't so much the maximum rated current but the lack of dynamic response.

For a jumping behavior for example all the motors will draw a high current at the same instance.
That can cause the voltage to drop if the power supply isn't dynamic enough to supply that current peak.

If the voltage at the motor drivers drops below 8V the microcontrollers on the motor driver cards will reset.
So for the high jumps we were using two power supplies - one to power the front legs and one to power the hind legs.
https://www.youtube.com/watch?v=javHeKRKbAc

FranCOOlin, 2022-05-01:
> We use a 30A power supply - you can find more information here: [...]

Thank you. Now I've got basic ideas about it.

closed 2022-05-02 by fgrimminger