open-dynamic-robot-initiative/open_robot_actuator_hardware #49: details_actuator_module_testing.md is still empty,how can I test the module before all stacked together?

vxgu86, 2022-10-02:
like what I asked in the title, how can I test the module before all stacked together?

https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/actuator_module_v1/details/details_actuator_module_testing.md#details-actuator-module-testing

by the way, it's really nice job.

thomasfla, 2022-10-03:
Hi, you can connect the module to the masterboard and the masterboard to a PC and run this example: https://github.com/open-dynamic-robot-initiative/master-board/blob/master/sdk/master_board_sdk/example/example.py

It should move the motor following a sinusoidal motion and display all the data from the actuator (position, velocity, current, index detection etc..)

vxgu86, 2022-10-03:
thanks, I also want to ask another small question, it's about the code wheel in [this link](https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/actuator_module_v1/details/details_encoder_kit_disassembly.md)

*The aluminum collar on the code wheel has to be removed on a lathe.
Cover the optical disc with tape to protect from scratching.*
what do you mean by "removed on a lathe", should the aluminum collar be cutted off, or grind off?
and by the way, how to clamp the codewheel with your tool in [this link](https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/actuator_module_v1/details/details_encoder.md)?

fgrimminger, 2022-10-05:
We use a "parting-off tool" to cut off the collar on a lathe - there's a picture below. 

We clamp the code wheel on the smaller diameter of the tool using a M3 screw and washers.

The larger diameter of the tool is held by the chuck of the lathe.

Alternatively you can order the motor shaft assembly with code wheel from PWB - as described [here](https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/actuator_module_v1/actuator_module_v1.1.md#description-odri-encoder-kit). 

[image]

[image]

vxgu86, 2022-10-17:
@fgrimminger thanks for your reply.
I tried one,but get the wheel twisted， not flat as original, not tested yet, don't know whether it would affect the performance.
by the way I did get the meaning of the first pic.

fgrimminger, 2022-10-18:
Hi,
o.k. - it might still work if it's not twisted too badly.
The encoders are pretty robust to misalignment and inaccuracies.
Just make sure that the code wheel doesn't touch the encoder when it's spinning - otherwise the disc will get scratched.
The motor control firmware checks how many pulses have passed every time the index pulse is recognized.
If the discrepancy is more than a couple of ticks the card will stop and report an encoder error.

Do you mean you **didn't** get the meaning of the first picture?
The first picture shows a parting-off tool - similar to what we use to remove the aluminum flange on a lathe.
If you do a web search for "parting-off tool" you'll find more variants and information.