open-dynamic-robot-initiative/open_robot_actuator_hardware #50: Q: what's the total estimated cost of the hardware?

mw66, 2022-10-05:
Just want to get a rough idea.

Thanks.

fgrimminger, 2022-10-11:
The price of the robot is difficult to estimate precisely since it will depend on the suppliers and machine shops that you use, the country you are in and the part quantities that you are ordering.

We have estimated the material cost for our Solo12 robot at around 6300€ + tax.
That is not including the imu which is the most expensive part on the robot.

PAL Robotics offers a kit with all the parts and alternatively a preassembled robot.
You can request a quote here: https://solo.pal-robotics.com/solo

easybuildtech, 2022-10-14:
Hi Pascal, check out https://odri.discourse.group/categories for a discussion forum.

fpgirard, 2022-10-14:
@easybuildtech - this is perfect - thank you.   I'm a huge fan of the University of Maryland where many of my interns matriculate.   Some of them are at JPL now (Nick R and Camden S.).    I'm liking what UMCP is posting [here](https://odri.discourse.group/t/interested-in-making-a-modified-less-expensive-solo12/140) - it's consistent with my initial observations.

thomasfla, 2022-10-15:
Hi, thanks for your interest!
indeed the [forum](https://odri.discourse.group/) is the good place to discuss such topic.

I can start here with a few element of answer. 
Despite being mostly 3d printed, solo and other ODRI robots are not low cost. At least not optimized for cost.
Those robots are used as platform to experiment and develop state of the art research. For academia, it make much more sense to invest in quality sensor and cutting edge compact electronic that ease our life while experimenting. 

I can understand how this can be frustrating from a hobbyist point of view, but I think you can find all the resources on this GitHub and in the forum to replace some components, design different actuators, etc.  Actually, you will find in the forum some threads about lower cost alternative to the choices made in the project.

About specific question / suggestion:

- Yes a low cost IMU could be used, with a more complex state estimation and calibration, but I would still expect a decrease of performances.
- The master-board is a compact design. Using a shield and standard dev board is possible but would take more space, In Solo12 we have very few margin. The Ethernet is a must to do high frequency feedback loop with a computer.
- The motor could be different, but some CAD adjustment are needed to accommodate with different size, and fixture.
- The cost is also due to the custom electronic to control the motor, Other lower cost project exist, but are not as compact nor adapted to our need.
 
Please understand that we would love to see spin-off of this project optimizing cost, exploring different design choices but we can not invest our work time to do it our-self.

I hope this helps,
Thomas.

MEE2022, 2022-11-08:
What is the estimated cost of the biped hardware (if you don't count the 3d printed parts)?

easybuildtech, 2022-11-24:
@MEE2022 I have been noting costs, on a BOM, as I build if it helps:  https://easybuildtech.com/open-dynamic-robot-initiative-bolt-cost-breakdown/

I am in the U.S.

MEE2022, 2022-11-24:
> <open-dynamic-robot-initiative/open_robot_actuator_hardware/issues/50/1326578748
> @github.com>

Thank you very much.  This is very detailed and extremely helpful.

fgrimminger, 2022-12-06:
@easybuildtech 
Great! Thanks for sharing!

closed 2022-12-06 by fgrimminger