open-dynamic-robot-initiative/open_robot_actuator_hardware #55: Foot Sensor Voltage supply

Seerzy, 2023-06-09:
Hi Guys,

I am using the same feet design from open dynamic. It is very good, and perform very well.
However, my system only have 5V, can I use the same design but change the supply voltage from 3v3 to 5V.
I did some testing, 5V supply can work for short term, but I do not know whether it gonna affect the life span or not.

Tom

Seerzy, 2023-06-13:
Hi Guys,

I though it gonna be a problem coz the voltage of the LED sensor is meant to be 3.3V. Can someone help me verify whether I can use 5V to power the feet sensor system?

fgrimminger, 2023-06-13:
Hi Tom,

I'm glad to hear that the foot sensor works for your project.
I talked to Joshi Walzog, who had designed the board, regarding your question.

His only concern were the external indicator leds which get too much current at 5V and might get damaged over time (about 60mA instead of about 20mA).
So he recommends to increase the resistance of the resistors R1, R2, R6 and R7 to 150ohms (51ohms at the moment for 3.3V).
For keeping the light intensity of the internal led at the same level you could increase the resistor R8 to 4k ohms (560ohms at the moment for 3.3V) - that's optional and will depend on how you want to tune the sensor characteristics.

We designed the board for 3.3V because the ADC on our microcontroller can only handle up to 3.3V.
When you increase the voltage to 5V you need to make sure that your microcontroller can handle 5V on the inputs.

Below is the latest schematic for reference.

I hope that helps.
Best wishes. Felix

[image]

eijijiokk, 2023-06-27:
Dear Felix Grimminger,
Thank you for share your research work with the community. I would like to know if is possible to get the STEP files for the design of the actuator module shell cover, since there is an issue  to convert STL file to a mesh. It missed some parts of the design.
Thank you so much.

fgrimminger, 2023-06-27:
Hi,
sure - below is the upper leg shell base and cover in the STEP format.
Best. Felix

[upper_leg_module_step.zip](https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/files/11884271/upper_leg_module_step.zip)

fgrimminger, 2023-07-24:
Dear Eliana,
you're welcome!
Thank you and good luck with your project!
Best. Felix