open-dynamic-robot-initiative/open_robot_actuator_hardware #56: Dear my friends, I have some questions about your electronics part.

wangyear, 2023-07-10:
Excuse me, could you tell me what the software you have used to design your Micro Driver Electronics (pcb), looking forward to your replay. thank you very much.

nim65s, 2023-07-10:
Hi,

Our latest project use kicad, you'll find it here: https://github.com/open-dynamic-robot-initiative/power-board/tree/main/hardware

wangyear, 2023-07-10:
> Hi,
> 
> Our latest project use kicad, you'll find it here: https://github.com/open-dynamic-robot-initiative/power-board/tree/main/hardware
Thank you very much, I will try it.

wangyear, 2023-07-20:
hi，please sign up your github account，i have some quessions about your project.Sorry to bother you.
---- Replied Message ----
| From | Guilhem ***@***.***> |
| Date | 07/10/2023 23:19 |
| To | open-dynamic-robot-initiative/open_robot_actuator_hardware ***@***.***> |
| Cc | wangyear ***@***.***>,
Author ***@***.***> |
| Subject | Re: [open-dynamic-robot-initiative/open_robot_actuator_hardware] Dear my friends, I have some questions about your electronics part. (Issue #56) |

Hi,

Our latest project use kicad, you'll find it here: https://github.com/open-dynamic-robot-initiative/power-board/tree/main/hardware