open-dynamic-robot-initiative/open_robot_actuator_hardware #57: Dear my friends, i have some questions about your sensors used in your project.

wangyear, 2023-07-17:
I'm sorry to bother you to tell me that what the sensors you have used in your project, and its concrete information. Now i just found only one sensor used in your wheels, right? If you have used other sensors, please offer me the information. Thanks a lot, looking forward to your reply.

fgrimminger, 2023-07-24:
Hi,
for measuring the joint positions we use an optical encoder:
https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/actuator_module_v1/README.md#optical-encoder

The only other sensor that we are using is an IMU on the body of the robot:
https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/quadruped_robot_12dof_v1/README.md#inertia-measurement-unit

Best. Felix

wangyear, 2023-07-24:
Thank you so such.
---- Replied Message ----
| From | Felix ***@***.***> |
| Date | 07/24/2023 19:47 |
| To | ***@***.***> |
| Cc | WANGYI ***@***.***>,
Author ***@***.***> |
| Subject | Re: [open-dynamic-robot-initiative/open_robot_actuator_hardware] Dear my friends, i have some questions about your sensors used in your project. (Issue #57) |

Hi,
for measuring the joint positions we use an optical encoder:
https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/actuator_module_v1/README.md#optical-encoder

The only other sensor that we are using is an IMU on the body of the robot:
https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/quadruped_robot_12dof_v1/README.md#inertia-measurement-unit

Best. Felix

fgrimminger, 2023-07-24:
You're welcome.
Thanks for your interest.
Best. Felix

closed 2023-07-24 by fgrimminger